nordicsemi/IOS-nRF-Connect #108: Want my app in English

Alfredodelaquintana, 2022-03-23:
Dear sirs
When download the app nfr connect for mobile; the app appears in Spanish language in my cel.
The translations are not always good, especially when translating technical terms.
How can I use the app in English?
thanks in advance

dinesharjani, 2022-03-28:
@Alfredodelaquintana yes, sorry. This is a holdover issue from the past when we thought about adding multiple languages.

The current version 2.5 solves this by removing all other languages. If you send me your email address via the 'Send Feedback' feature I can add you to the beta.

Alfredodelaquintana, 2022-03-28:
Dear Dinesh:
Thanks for your answer.
My email is ***@***.***
Can´t send the email using the option that you mentioned because the
version that I'm using is in spanish and the option "send feedback"
("enviar realimentación" in spanish) don´t exist.

Waiting for your new instructions.....

Regards

         Alfredo De La Quintana G.
             CONECTA Ingeniería
         Dr. Alejandro del Río 2171
               Ñuñoa - Santiago
             Fijo: +56 2 699 7913
           Móvil: +56 9 934 38482

dinesharjani, 2022-03-28:
Your email is not visible :)

Write to me at dinesh.harjani@nordicsemi.no.

closed 2022-05-17 by dinesharjani