nordicsemi/IOS-nRF-Connect #109: nRF requesting access to bluetooth when closed

GitItV, 2022-04-14:
iOS 15.4.1 iPhone

iOS dialogue box appears with nRF prompt to connect to Bluetooth randomly. 

app is force closed
Background app refresh is off
Cell permission is off

wat?

-please be kind. I have no idea what I’m doing. I’m new to this.

What am I doing? Learning how certain Bluetooth devices work so I can use them safely.

iPhpwn4, 2022-07-02:
I'm experiencing the same behaviour. 
iOS 15.2
iPhone XR

WinkelCode, 2026-09-09:
Same problem here, even taking away its Bluetooth permissions doesn't seem to stop it. Really annoying.

Other apps like LightBlue don't seem to have this issue.

dinesharjani, 2026-09-10:
@WinkelCode I imagine this happens when you've not accepted / allowed the app to have Bluetooth permissions?

WinkelCode, 2026-09-10:
@dinesharjani 

Context: I disable Bluetooth on my iPad when not in use (I don't have any BT accessories I regularly use).

I re-installed nRF Connect recently to troubleshoot a BT accessory, after I was done (device was definitely unpaired/forgotten at that point; I am also 99% sure it has been properly off during times I got the notification) I closed the app and moved on with other stuff. Some time later I got the prompt - while in either another app or the home screen - that I should turn on Bluetooth so nRF Connect (called out by name in the prompt) can use it. This happened a few times, I then checked if it was allowed as a Background App (not listed) and removed its Bluetooth permission, which ultimately did not fix it as I kept getting the same message to turn on Bluetooth every so often. Based on behavior when opening the app, it appears the OS always prompts to turn Bluetooth on first before any prompt regarding permissions would appear.

Note: I am on the iPad OS 27 Beta, however, LightBlue (similar BLE testing app which I also used to connect to the aforementioned accessory) doesn't exhibit this behavior, and the issue here is some years old already.

dinesharjani, 2026-09-16:
Sorry for the long delay, I stepped out to a conference. This is very good and helpful feedback. With the key detail of having the permission off / disabled, I can hopefully replicate this and fix the issue properly. I'll get back to you as soon as I'm able to.

Thank you.