nordicsemi/IOS-nRF-Connect #116: Descriptors Not Found

coalbr, 2022-09-15:
I have a custom BLE device that has custom characteristics and descriptors implemented.

Using version 4.25.4 of the Android application I am able to read the characteristics and their descriptors without issue.

Using version 2.5.3 of the iOS application I am able to read the characteristics without issue but the descriptors are not found. I can even see in the logs that it indicates no descriptors are found for that characteristic.

philips77, 2022-09-16:
Could you share the UUIDs of the descriptors? Perhaps iOS is hiding them from the app. By having UUIDs we could try to replicate the issue and check it, Thanks!

coalbr, 2022-09-16:
@philips77 So I have a characteristic with a UUID of 5642**0401**-a454-11eb-bcbc-0242ac130002 and that characteristic has descriptors with UUID values of:

- 5642**ff01**-a454-11eb-bcbc-0242ac130002
- 5642**ff02**-a454-11eb-bcbc-0242ac130002
- 5642**ff03**-a454-11eb-bcbc-0242ac130002
- 5642**ff04**-a454-11eb-bcbc-0242ac130002
- 5642**ff05**-a454-11eb-bcbc-0242ac130002

coalbr, 2022-11-09:
@philips77 Are there any updates on this?

philips77, 2022-11-10:
@dinesharjani Could you look into it?

dinesharjani, 2022-11-10:
So I tested setting up a GATT Server with your UUIDs, and it was scanned properly. I wonder if your issue is because you've programmatically changed the Services of your custom BLE Device, and iOS aggresively caches this. You can try to go to Settings -> Bluetooth and turn it off and on; doing so from Control Center doesn't clear this cache, it just stops BLE Advertising.

nimpy, 2023-03-31:
I also have this issue -- for a custom BLE device with custom service, custom characteristics and custom descriptors (as defined using [google bumble library](https://github.com/google/bumble)), on Android I can see the descriptors, whereas on iOS I can't.

There are also other discrepancies between iOS and Android when it comes to these attributes:
- When I don't add any descriptor for a characteristic, there is still the CCCD descriptor (UUID 2902), which on iOS is readable and writable, but on Android only writable (I'm saying this based on the options that nRF Connect is showing)
- When I add Service with [GATT_GENERIC_ACCESS_SERVICE](https://github.com/google/bumble/blob/45ca0ef071425f3c11cb57dc2ae95c9a0f2df483/bumble/gatt.py#L54) UUID, which has a Characteristic with [GATT_DEVICE_NAME_CHARACTERISTIC](https://github.com/google/bumble/blob/45ca0ef071425f3c11cb57dc2ae95c9a0f2df483/bumble/gatt.py#L168) UUID, on Android I can see this characteristic, whereas on iOS I don't see any characteristics within this service.

I believe this is _not_ an issue with nRF Connect app but with iOS, because another (custom) app also doesn't see the descriptors. I found this issue on iOS 15.1.1 and on 16.3.1, these are the only two I checked.

My question is: does anyone have any idea about this and how to deal with these discrepancies? Most specifically, I would like to be able to see added descriptors on iOS (like they can be seen on Android).

dinesharjani, 2023-04-03:
I'm going to try and add as much light as possible. I need to revisit all of this code soon(™) so perhaps there are opportunities for me to improve this.

Uhm, so the 2902. Yes, this is automatic behaviour from iOS and there's nothing we, on the developer side, can do about it. That's why we add it as 'locked' when you make a Server, if memory serves. I think I'm right (that we add it and mark it as locked in the UI.)

iOS Bluetooth heavily caches data. So if a Device changes quickly the Connected Services, we on the app side might get the data for the cached - incorrect Services. What we do during Development is go to Settings -> Bluetooth and turn BLE Off and On. This clears the cache, and be careful NOT to do this using the Control Center Panel, because doing this doesn't turn off BLE just stops Advertising and doesn't clear the cache. There's nothing we can do about this here either, we just try to keep up and notice the changes iOS makes.

I don't know if the Device Name Characteristic is not visible to you due to this caching or, if it's because of the filtering iOS does. For example, the Covid BLE Service - I don't remember the name now, 'Exposure Service' was it? - is filtered in iOS. But when running the exact same code, nRF Connect for iOS, on the Mac (yes, we plan to release this one day), we can see it. iOS does things like this in the name of protecting user privacy, trying to make it difficult to track people, etc. Also keep in mind iOS devices continually advertise their encrypted keys for 'Find My' Service as well as relaying keys that they find outdoors so AirTags work and so on, which takes away some available BLE bandwidth, though iOS apps appear to be unaffected. But I mention this so you're aware of all the kinds of shenanigans Apple does with Bluetooth on iOS.

nimpy, 2023-04-04:
Thank you for the response and for these insights! Good to keep that in mind :)

Regarding custom descriptors not showing (this is the main problem I would like to solve -- the other discrepancies are not impacting me for the moment :D), I don't think this is a caching issue since they weren't showing the first time that I connected to the device (and they should have been there).

I am curious to know: @coalbr, were you by any chance also using the [bumble library](https://github.com/google/bumble/) when you noticed this issue?

closed 2024-12-12 by dinesharjani