nordicsemi/IOS-nRF-Connect #118: DFU with NCS MCUBoot fails sometimes

trueb2, 2022-10-24:
I have experienced for several versions of both the NRF Connect iOS app and NRF Connect SDK an issue where NRF Connect iOS cannot OTA DFU with MCUBoot. I am opening an issue here because it does not seem to be an issue with the NRF Connect SDK. The NRF Toolbox iOS app succeeds to performing the OTA DFU update procedure. Both apps will sometimes timeout during sending the binary and succeed after retry when that happens. Specifically, the issue at hand is that NRF Connect iOS may not be able to initiate OTA DFU for the same sensor that NRF Toolbox iOS can.

One screenshot shows a failure on NRF Connect iOS that eventually results in a timeout (even after reattempt). The other screenshot shows the beginning of the logs from a NRF Toolbox success.

[image]
[image]

dinesharjani, 2022-10-25:
Hi @trueb2 - the version of nRF Connect on the App Store has an 'old' internal version of the DFU Library for OTA Updates. We're working as fast as we can on the new version, 2.6, but it's pretty big. What we can do in the meantime is I can add you to the TestFlight beta builds of it so you can check it out and let us know if the issue persists.

trueb2, 2022-10-25:
@dinesharjani That would be great!

dinesharjani, 2022-10-25:
@trueb2 Write me at dinesh.harjani@nordicsemi.no with the email to add to our TestFlight Group. That's all :)

closed 2023-02-16 by dinesharjani