nordicsemi/IOS-nRF-Connect #119: Device reports unknown error

DG12, 2022-11-16:
Your device reports an unknown error for Bluetooth use. 

Adding error code might be helpful. 

Tapping anywhere has no response. 
Must kill app. No change to any setting. 
Restart now ok.

dinesharjani, 2022-11-16:
Can you provide a screenshot?

closed 2024-03-05 by dinesharjani

DG12, 2024-03-05:
Would you prefer feedback email to you?  
to [mag@NordicSemi.com](mailto:mag@NordicSemi.com)?  or github issue?

dinesharjani, 2024-03-06:
Email or this issue works. I'd just like to get some more context because otherwise I don't know where to look for the issue - is Bluetooth disabled from Settings? Or from the Control Center? Are you using Peripheral Tab to Advertise or act as a BLE Server? Also, a screenshot would be helpful as mentioned.