nordicsemi/IOS-nRF-Connect #121: How to specify custom advertising service?

akohlsmith, 2022-12-07:
I can specify custom uuids for server services, but I cannot do so for the advertising part of the virtual peripheral. V2.5.3 build 4 on iOS.

rego21, 2023-05-01:
Hey,

Im also trying to do it. You can add a custom server with a service's uuid of 128 bits, however you cannot add an advertiser with a custom uuid only the pre defined ones.

thanks!

dinesharjani, 2023-05-02:
You know the app better than I do :)

Marking as closed.

closed 2023-05-02 by dinesharjani

rego21, 2023-05-02:
Im curious if the same happens in Android, sadly I don't have one to test :(

philips77, 2023-05-02:
Hi, on Android it's possible to set UUIDs for services and in Advertising. As the API is more flexible, you should also be able to set the Service Data, Manufacturer data and some other advertising parameters.