nordicsemi/IOS-nRF-Connect #124: Max. RSSI filter catches peripherals with invalid RSSI values

reggian, 2023-01-17:
In the scanner if one sets the filter for Max. RSSI (e.g. -40 dBm) the peripherals that have an invalid RSSI value (in the UI it is listed as `N/A`, I presume that it caught the RSSI value of 127) pass through and then stay on the list even though the later valid values are way below (e.g. -85 dBm) the set Max RSSI value.

dinesharjani, 2023-01-17:
Yes, you are correct. We made an improvement to this for the upcoming version 2.6 release of nRF Connect. You can try it out in the TestFlight beta if you write me back using the App's Settings -> Send Feedback option.

Thank you for raising the issue ^^

reggian, 2023-01-17:
Kudos @dinesharjani for such a fast response. I have sent you a feedback.

dinesharjani, 2023-02-16:
Marking as closed.

closed 2023-02-16 by dinesharjani