nordicsemi/IOS-nRF-Connect #126: Device Type field not complying to BTSIG "Appearance" definition

chrislro2019, 2023-02-16:
Our device publishes the Appearance as 0x0D44, but the latest version 2.6b616 (and versions before it) shows `Device Type: GlucoseMeter` although the Online Resources Status shows "Up To Date". This does not comply to https://specificationrefs.bluetooth.com/assigned-values/Appearance%20Values.pdf. At least the Category as defined in the PDF should be used (and maybe additionally display the hex value somewhere)

dinesharjani, 2023-02-17:
Very good point.

We're processing 'Appearance' on our own based on the Advertised Services of the Device, not on the published Appearance value. We will take this into consideration. Thank you.

philips77, 2023-02-22:
That information isn't available on iOS. The only data types available are listed here: https://developer.apple.com/documentation/corebluetooth/cbcentralmanagerdelegate/advertisement_data_retrieval_keys

mervyd1993, 2025-04-04:
> That information isn't available on iOS. The only data types available are listed here: https://developer.apple.com/documentation/corebluetooth/cbcentralmanagerdelegate/advertisement_data_retrieval_keys

mervyd1993, 2025-04-04:
> Our device publishes the Appearance as 0x0D44, but the latest version 2.6b616 (and versions before it) shows `Device Type: GlucoseMeter` although the Online Resources Status shows "Up To Date". This [...]

dinesharjani, 2025-04-15:
Hi,

nRF Connect has not yet adopted the Bluetooth SIG Appearance(s). It's still all-custom Appearance(s). Online Resources is mostly for Company IDs and GATT Attributes.

closed 2025-04-15 by dinesharjani