nordicsemi/IOS-nRF-Connect #129: URL in Eddystone format not shown correctly

ladivin, 2023-04-19:
There's a tiny bug in displaying Eddystone data. The URL is missing the "." sign in front of the domain. Instead of `www.zephyrproject.org` the application shows `www.zephyrprojectorg`. Please refer to the attached screenshot. 

Source of Eddystone data: Eddystone sample from Zephyr v3.3.0. 
iOS version: 16.4.1
nRF Connect version: 2.6.3

[image]

dinesharjani, 2023-04-20:
I hear you. I need to check a couple of things and come back to you. Thanks for bringing this up.

dinesharjani, 2023-04-20:
You are correct. I wanted to check the hex parsing code properly instead of looking at the internal code mapping. But yes, it was a pretty simple bug. Thanks for pointing this out.

Will ship in version 2.6.4.

closed 2023-04-20 by dinesharjani

ladivin, 2023-04-24:
Thank you for taking care of this!

dinesharjani, 2023-04-26:
If only everything else was this easy to fix! :)