nordicsemi/IOS-nRF-Connect #134: App crashes when trying to add zip file for dfu update for nrf5340

reddwarfSiddis, 2023-09-20:
Have tried this on multiple iPhone, from 12 - 14. All barfs at me when trying to upload dfu_application.zip file generated from nrf connect sdk(version 2.4.1).

dinesharjani, 2023-09-20:
Any specific sample code from nrf connect sdk 2.4.1 that I can generate a dfu_application.zip file?

I have a feeling I fixed this. I'm working on a bit 2.7 update, and it has a tons of fixes as you can imagine. But it's a good idea to double check.

reddwarfSiddis, 2023-09-22:
I think you need to add mcuboot and the SMT service to get dfu_application.zip generated.
The best example I have seen for this has been created by Sigurd Hellesvik/Nordic Semiconductor. He has collected some examples here -> https://github.com/hellesvik-nordic/samples_for_nrf_connect_sdk/tree/main/bootloader_samples

reddwarfSiddis, 2023-09-24:
Just to clarify another thing: The nRF Connect for Mobile for Android does work perfectly.

dinesharjani, 2024-01-03:
Ouch. Sorry, I slipped big time on this. Let me see if I can catch up on it today.

dinesharjani, 2026-08-21:
Closing since we haven't heard back.

closed 2026-08-21 by dinesharjani