nordicsemi/IOS-nRF-Connect #136: DFU unavailable on app version 2.7 (menu bar hidden on connected device)

carlvrj, 2023-11-22:
The option menu on the connected DFU unit have disappeared in the iOS app 2.7. Once disconnected you can see the option bar again. See screenshot.
Same issue on both iOS16.7 and iOS17.1
[image]
[image]

dinesharjani, 2023-11-22:
Checking now. Thank you very much for the feedback

carlvrj, 2023-11-22:
Hi, thanks for quick resolution. Yes, it would be nice to join your TestFlight beta program.

dinesharjani, 2023-11-22:
Send an email through the app. Settings -> Send Feedback and I can add you. The fix will take a couple of days to go live, maybe even next week, since the US is on Thanksgiving.