nordicsemi/IOS-nRF-Connect #140: Unable to load Firmware to begin update when using file on OneDrive

markb-EE, 2023-12-16:
When I try to use a file from Microsoft OneDrive I get the following error:

[image]

Steps:
* Add OneDrive app from Apple App Store to iPhone
* Upload a DFU .zip image to OneDrive account
* Open nRF Connect for iOS
* Connect to a device with DFU Service running
* Press the "Open Document Picker" button
* Browse to top level using the "< Back"/"< Browse" buttons
* Select the OneDrive location (see image below)
* Select DFU .zip image
* Screen with error above is displayed

[image]

ChaunceyLong, 2023-12-26:
This happened to me too, how did you solve it?

dinesharjani, 2023-12-26:
Very dumb question - (sorry for the delay since I'm vacation until next week) - do you know the .ZIP file is a valid file for DFU?

Otherwise, it could be a parsing error, or an issue reading the file specifically from OneDrive. You can also import the file into nRF Connect by connecting the iPhone to the Mac and dragging it into nRF Connect from the Finder. You can also try to read the file from iCloud Drive.

markb-EE, 2024-01-03:
The .zip file is valid. I’m able to us it for DFU if I import it to nRF
connect. I did this by using the “Open in Another App” option in One Drive
and selecting nRF Connect:

This has been a long standing issue, I only just got around to reporting it.

dinesharjani, 2024-01-03:
Disregard my previous comment, I'm able to reproduce the issue using OneDrive as you mentioned.

dinesharjani, 2024-01-04:
Alright, I believe I have this fixed. It will ship with 2.7.3 as well.

closed 2024-01-04 by dinesharjani