nordicsemi/IOS-nRF-Connect #145: updated IOS nRF connect app does not show the expected returning data from firmware, see screenshot for details

xingguo-huang2, 2024-01-05:
I was using nRF connect IOS version just fine, but someday a few week ago, since I updated the app on my iphone, then the app stopped working as expected(no return data from firmware can be observed on the app). 

I happened to have an older version of the **iPad** IOS nRF app which works fine. 

However, when I try to use my updated **iPhone** app to communicate with the firmware, it won't show me the returned data.

[image]
[image]

dinesharjani, 2024-01-08:
Oh, that's a good one. So what's happening is that the 'write' is not triggering a Log update, but the Write is happening as you can see from 'Client' tab. It's a fault caused by optimising the Logs screen to not update in the background when you can't see the logs, only it's not updating when you return to them.

dinesharjani, 2024-01-08:
Marking this as fixed. Will ship with version 2.7.3.

closed 2024-01-08 by dinesharjani

xingguo-huang2, 2024-01-26:
> Marking this as fixed. Will ship with version 2.7.3.

Upgraded to 2.7.5, the problem is still not fixed.....

[image]
[image]

xingguo-huang2, 2024-01-26:
Please help fix the above issue... Thank you!

philips77, 2024-01-26:
Based on the image from the first comment you're disabling notifications just after enabling them:
[image]

When disabled, you'll not get any updates.

xingguo-huang2, 2024-01-26:
[image]
Thank you very much! This worked for me. Just post it here in case someone will need it.