nordicsemi/IOS-nRF-Connect #146: iOS App crashes on "start" press for DFU

alexheslop, 2024-01-22:
I'm connecting to an nRF52832 chip using the iOS nRF Connect app, and trying to trigger a DFU. When I load the file and press start to kick off the transfer, the application crashes and resets.

reddwarfSiddis, 2024-01-23:
Same with nRF5340. Worked great in last release

dinesharjani, 2024-01-24:
@reddwarfSiddis So for the n5340, I believe it's a crash we tried to fix, but didn't fix, and I did not merge it properly for the newest release. But we have a fix for it. I'm trying to add some more fixes related to iPad use of the app and hopefully I can send it to Apple for release this week. Sorry for the inconvenience.

dinesharjani, 2024-01-25:
Ok, the nRF5340 should be fixed in 2.7.5, which is in the App Store already.
@alexheslop - what SDK version is the DFU? I presume this is 'NordicDFU' and not the newer Zephyr / McuMgr, correct?

philips77, 2024-01-25:
Could you paste here the *manifest.json* file from the ZIP which you're trying to upload?

alexheslop, 2024-01-25:
{
    "format-version": 0,
    "time": 1706197950,
    "files": [
        {
            "load_address": 49664,
            "version_MCUBOOT": "0.0.0+0",
            "type": "application",
            "board": "development_kit",
            "soc": "nRF52832_QFAA",
            "size": 229664,
            "file": "app_update.bin",
            "modtime": 1706197950
        }
    ],
    "name": "************************************",
    "firmware": {
        "zephyr": {
            "revision": "91eb13d3afd7ddc7dacfb7435a93838392bd714b-dirty"
        },
        "nrf": {
            "revision": "75adfe8909fbac1c2e0ede465448d1ce2821dfa9-dirty"
        }
    }
}

dinesharjani, 2024-01-25:
Let me try this out tomorrow. I think I see a possible issue when parsing the manifest that I might need to fix. Thanks.

alexheslop, 2024-01-26:
Thanks @dinesharjani!

dinesharjani, 2024-01-26:
@alexheslop something weird is going on during Unzip. It's not doing what I expect. I imagine when you select the file, the Preview is not populated?

alexheslop, 2024-01-26:
[image]
[image]

alexheslop, 2024-01-26:
It seems to preview it okay @dinesharjani, but it fails instantly on sending the request

dinesharjani, 2024-01-26:
Okay. But it doesn't crash.

You can swipe left to the Logs, enable all Levels, and get more info about what happened.

closed 2024-12-12 by dinesharjani