nordicsemi/IOS-nRF-Connect #147: Button for "Write request" in a BLE service not responding

carljohanlandin, 2024-02-20:
I have a BLE application based on the peripheral_lbs sample running on a nrf52840dk. I have previously been able to use the iOS net connect app to send write requests to my device but after the app has been updated the write button in the BLE service is unresponsive.

I have confirmed that the problem is not in my application by using net connect desktop, from which it works sending the same write requests.

dinesharjani, 2024-02-20:
I will make a fresh peripheral_lbs sample and run it on an nrf52840dk - thanks for specifying that info.

But, if you move over to Logs, can you see the write requests happening? Remember to enable all Debug Log Levels to check for this.

carljohanlandin, 2024-02-20:
Hi Dinesh, Thanks for looking into it.

I have enabled the debug log level as well but still can’t see any log
events related to writing after pushing the write button.

Best regards
Carl-Johan

dinesharjani, 2024-02-20:
Which iPhone are we talking about?

carljohanlandin, 2024-02-20:
iPhone 12 Pro

dinesharjani, 2024-02-20:
May I see a screenshot? Perhaps this allows me to have a better idea of what the issue might be. A screenshot of Device Details (Client), I assume this is where the issue is.

dinesharjani, 2024-02-20:
Unfortunately I can't see the screenshot. Neither via email nor GitHub comments :(

You may cover up the names / UUIDs of Services and so on. I just need to get a general idea of the state of the UI, and see the Buttons and in which cells / attributes they're visible / usable from.

carljohanlandin, 2024-02-20:
[image]

Hope this works better

dinesharjani, 2024-02-20:
Definitely better but unfortunately I cant see the buttons which are on the right side of the screen.

carljohanlandin, 2024-02-20:
I realised I accidentally cropped the picture. Here it is again

[image]

dinesharjani, 2024-02-20:
I see. I will do some testing tomorrow. But this screenshot is helpful - thank you.

carljohanlandin, 2024-02-20:
Great, thanks for the help.

Best regards
Carl-Johan

dinesharjani, 2024-02-21:
When you tap the Write button (arrow pointing up), do you get the Write Characteristic Dialog? And you can type in Data and then tap 'Write' ?

carljohanlandin, 2024-02-21:
No, nothing happens. No dialog appears.

dinesharjani, 2024-02-21:
If you try to Write another Characteristic / another Device etc, does it work? Is it only for the particular Attribute / firmware you're writing or is it in general the Write Dialog doesn't show up for anything?

carljohanlandin, 2024-02-21:
Just now I tried to flash the actual peripheral_lbs sample (instead of my
own app built from it) and the same issue remains. However, I can interact
with the read buttons. Enabling notifications for when the button is
pressed and released works. Also reading the client charecteristic
configuration attribute works. But both the “Blinky LED state” write button
and the “Client characteristic configuration” write button is not
responding. I can only see those buttons changing color as I hold my finger
on them. But that’s it.

dinesharjani, 2024-02-21:
Ok, I believe I see the issue. It's related to the Write Type, which looks like it's registered as 'Authenticated Signed Write'. I managed to cause it and reproduce it in the current version out in the App Store - Write Button is there but nothing happens no matter how much I tap it.

Can I send you a TestFlight beta link with a possible fix to your email address ?

carljohanlandin, 2024-02-21:
Good catch:)

Sure, no problem.

dinesharjani, 2024-02-21:
@carljohanlandin I cannot see your emaiil because GitHub is protecting it.

Please write me to dinesh.harjani@nordicsemi.no with the email address of your Apple ID (or the Apple ID of the device where you want to test this) so I can add you to TestFlight

dinesharjani, 2024-02-22:
Closing as this issue has been confirmed by @carljohanlandin as fixed for nRF Connect 2.7.6 (in Beta)

closed 2024-02-22 by dinesharjani