nordicsemi/IOS-nRF-Connect #148: iOS Peripheral tab alignment issue

janwiesemann, 2024-02-21:
Hello!

I’ve recently started my journey into the BLE world. For this I’m using your app and noticed a very annoying issue on the Peripherals Tab. If you enter a Service UUID on iOS and close it, the peripherals header bar is being moved down and only a small segment is actually visible.

I’ve uploaded a video to highlight the issue: https://youtube.com/shorts/yBJC0_X3Eg4?si=r-kWwHhm3ftLs438

dinesharjani, 2024-02-22:
I can reproduce the issue on the iPhone 15 Simulator. Thank you, the video was very helpful. I'll get on fixing this.

dinesharjani, 2024-02-23:
I managed to fix this. It's not 100% perfect, but it is fixed I think. Your video was really helpful. Fix will ship with version 2.7.6.

If you'd like to access the fix early, send me an email to dinesh.harjani@nordicsemi.no with the Apple ID you'd like to get the TestFlight invite for. In other words, the Apple ID of the device you'd like to run the beta.

closed 2024-02-23 by dinesharjani