nordicsemi/IOS-nRF-Connect #149: Modal Screen already visible - Alert prevents opening connected device

logsol, 2024-02-26:
I am able to scan for devices but when I connect to my peripheral, I cannot go to the services and characteristics. I see this alert view, that says: 

"Panel already open - There is already a Modal Screen visible to the user"

[image]

This did not happen before on my iPhone SE 2nd generation, but keeps happening on my iPhone 15. 

Any hints?

dinesharjani, 2024-02-26:
Easiest way to fix this is to go to Settings -> Version Number and tap it about 15 times until "Developer Mode" is enabled. When it is, scroll back up and do "Restart App". Should fix it.

And then, if you can next tell me a sequence of steps to follow to reproduce this on an iPhone 15, I can fix it.

Thanks.

dinesharjani, 2024-02-26:
I will have another look at the code tomorrow as well to see if I can get a glimpse of how this could happen. The iPhone 14/15s unfortunately have a couple of weird situations that prior iPhone(s) don't get into, I presume because of the Dynamic Island. Was this on a Max? Plus? Pro? Or regular iPhone 15?

logsol, 2024-02-26:
Oh wow thanks for the quick reply with the workaround. I’ll be sure to check this tomorrow. Unfortunately I don’t have any repro steps other than what I already wrote. However I’ll keep an eye out if it happens again and if there’s some pattern. I am using a regular iPhone 15.

dinesharjani, 2024-02-26:
Yeah, I don't have a 15. But already there have been a couple of issues with 14/15 so 16 is on my list now, though that's a few months away. The Simulator helps, but it doesn't work with Bluetooth so I have to mock the Data. Anyway - let me know tomorrow if this works for you. And if there's a sequence of steps to reproduce this, please let me know.

Version 2.7.6 was also just submitted to Apple. It has a lot of changes, I don't remember now if this kind of state management is involved or not, so I'm not expecting this to fix it.

dinesharjani, 2024-02-29:
Okay, I think I know the cause. It's something to do with rotation - device on landscape, with a device open. Switch tab to RSSI, go to portrait, back to Scanner and there's the Modal Alert. It's rotation logic at fault.

dinesharjani, 2024-03-05:
This is fixed and will ship with version 2.7.7.

closed 2024-03-05 by dinesharjani