nordicsemi/IOS-nRF-Connect #150: MacBook Silicon,  Sonoma needs SEGGER universal

DG12, 2024-03-11:
[image]

DG12, 2024-03-11:
Installed Segger Universal 796 now

nRFconnect complains:

```
2024-03-11T00:11:28.994Z INFO Using JLink version: 
JLink_V7.96 

2024-03-11T00:11:28.994Z WARN Installed JLink version 
does not match the expected version (JLink_V7.88j)

```

dinesharjani, 2024-03-11:
Hi Dennis, we appreciate your feedback. But this GitHub is about the iOS / Mac version(s) of nRF Connect. You're providing feedback about the general nRF Connect for Desktop. I recommend you share your feedback in our [DevZone forums](https://devzone.nordicsemi.com/). Meanwhile, I'll have to mark these issues as closed - sorry for the inconvenience.

closed 2024-03-11 by dinesharjani