nordicsemi/IOS-nRF-Connect #155: Beurer BF600 Device Reconnection Issue in iOS App

jatin141141, 2024-03-29:
We are currently in the process of developing an application(iOS) that integrates Beurer devices for measurements. However, we're encountering difficulty with the BF600 device reconnecting after initial pairing. Despite attempting to reconnect by forgetting the device from the Bluetooth settings on the iPhone, we find that it only connects successfully the first time. Subsequent attempts result in the same error. We've extensively checked using both the Beurer HealthManager Pro and NRF Connect apps, but unfortunately, we encounter the same issue in both applications. Please find attached links to videos showcasing the issue in both applications for your examination.

Could you assist with resolving this issue? We're currently stuck on integrating the BF600 device with our iOS application.

NRF Connect:
https://we.tl/t-tPwJErul7H

Please refer this link we are check your official ios app still getting issue for second time connecting.

dinesharjani, 2024-04-01:
So, first of all, you're very smart in going to Settings, Bluetooth and forgetting the Device.

My suspicion is that the issue you're encoutnering is that both the iPhone (not the apps) and the Beaurer fail to negotiate. Pairing on iOS essentially means "encrypted" communication. That's why the pairing request only shows up after you try to send Data. By going to Settings and Forgetting, you're clearing the Cache of the previous pairing certificate on the iPhone. You may also try to Forget the Device, turn off Bluetooth from Settings (not from Control Center), and then turn it on again and try. This wipes the local cache on the iPhone. But then the issue might be that the iPhone is not forgotten on the side of the Beurer BF600, so it's trying to use the previously negotiated pairing key, which the iPhone will reject, and then the connection fails. 

Best course of action to understand what's going on is to use a BLE Sniffer. You may [use an iPhone as a sniffer](https://www.bluetooth.com/blog/a-new-way-to-debug-iosbluetooth-applications/), if you have a Mac, or some other commercial device / software. But the Sniffer will provide you with the most information regarding why is the connection really breaking down. There's unfortunately not much else I can do. And if I'm correct and the Beurer is the one trying to communicate with the old pairing key, unless you can reset that, this looks like it won't have a fix.

jatin141141, 2024-04-01:
Hello,
We have one query regarding in first time BF600 is connected successfully and get the data but second time didn't get the reconnect.

This error will come when second time connect "peer removed pairing information ios"  so in that case we have manually remove device from Bluetooth in setting and connect again. then it will work.

Can you please provide how to resolve the issue from second time reconnect the device

dinesharjani, 2024-04-01:
Then that sounds more like the BF600 is not reusing the pairing key from the first connection, and is treating each connection like a new one. There's no way to tell this to iOS from iOS apps.

closed 2024-12-12 by dinesharjani