nordicsemi/IOS-nRF-Connect #156: log on-screen should show hh:mm:ss.mmm  - improvement

DG12, 2024-04-02:
The log on-screen currently shows entries like:
```
Scanner on.     3 min. ago
Device Scanned. 3 min. ago
Scanner Off.    1 min. ago
```
1 Granularity in minutes is too large.
2 Looking at the screen now (i.e. 12 minutes later ) I have no idea when events occurred.
I would prefer a format like:
```
Scanner Off.         14:23:59.574 
Device Scanned.      14:23:58.893
Scanner On.          14:23:58.321
```
3 Less space between events would permit more events to be seen without scrolling.
4 Shorting the description would permit more events to be seen (especially on iPhone.
( "String" could always be omitted). 
Perhaps:
```
 Information                   Info
 Revision                      Rev
 Discovered  Device            Device
 Nordic UART Service Services  UART
 Characteristics               " "
 has no descriptors.           <NULL>
 Manufacturer Name             Manufacturer
 
 Device appearance switched    Appearance now UART
    from Generic to UART 
```

Most recent event inserted at top of list would be helpful.
[image]

Thank you for your consideration.

dinesharjani, 2024-04-04:
So, I will make changes to the text to shorten it. However, on showing the timestamps on the logs... I'll think about it. Logs can get very crazy when you do things like DFU. And you might do multiple DFU attempts, and when you do that, so many numbers (bytes) plus the timestamps as numbers, it gets confusing in my opinion. So reading "when" something happened rather than having to process in your head "was this the attempt 5 minutes ago or the one half an hour ago" by calculating current time - timestmap, is not a black & white tradeoff. But I will reduce some logs based on your suggestions.

dinesharjani, 2024-07-16:
I'm not sure if there's anything more to do here, then. Please raise another issue or comment again if there is.

DG12, 2024-07-18:
Is there a release available for this? Thank you

dinesharjani, 2024-07-18:
The problem with this issue is that no one will ever be happy. If I switch the logs to show timestamps, like I did for when we export Logs, then everything is a lot of numbers on the right side of the screen that look too close to each other. Also, since the exact timestamp will be wider, there would be less space for the "text" of each log line as well. As for less space between events, I guess I could see if making the paddings smaller works, but again, you need space between the text lines or else it gets all muddied up pretty quickly. What I would suggest is that, you tell me alternatively if the current Log Levels / Log Filter combinations don't work for you, or what I could I do to make them better for your use case.

dinesharjani, 2024-07-18:
I can work on the descriptions you proposed in point 4. Sorry, I confused your update with the "timestamps in the logs" to understand you were happy with that change.

DG12, 2024-07-19:
Thank you very much. 
Some of the misunderstanding is undoubtedly my fault.
( Some may also be the assistance of "spelling"correction).
When I first entered this issue I had not seen to export of the log and added that comment later!

Please let me know when you have something new.

PS My old iPad 16GB has IOS version 9.3.5 and nRF-Connect is not compatible! I a using the much smaller iPhone. Notice screen print on original posting :-) Maybe I'll try to find  another iPad on eBay.

I don't know how many users have iPads.

dinesharjani, 2025-01-07:
Hello,

There was a change made in 2.7.14 which now shows the log timestamps in a small bubble in a more traditional format. As of this moment, there are no plans to "go back" so, this should be stable. Or at least given the option to remain in the future. I think this marks the issue as closed.

closed 2025-01-07 by dinesharjani