nordicsemi/IOS-nRF-Connect #159: Uploading an image that uses external SPI flash throws “Error: Not Supported”

zpm1066, 2024-04-19:
Hello,
I’m seeing a potential bug when uploading an image to the external flash of a device.

After the upload, I get the “Error: Not supported” and device disconnet. However, FOTA BLE image upload works fine.

Uploading an image that uses the SoC flash is fine. After the upload, nRF Connect waits for device to swap partitions and reboot with the new image, and messages states “Sucess!”.

Please review and provide a bug fix.

nRF Connect is a great app. Keep it up! Thank you.

philips77, 2024-04-19:
I think this applies to "Erase app storage" command. It doens't matter for the success of upgrading the device. 
You may also disable erasing storage.

philips77, 2024-04-19:
Which upload mode are you using? Are you using Direct XIP as well?

zpm1066, 2024-04-19:
I’m using “DFU” from nRF Connect for mobile iOS and using the defaults.

closed 2024-07-16 by dinesharjani