nordicsemi/IOS-nRF-Connect #160: Enhancement request - Add custom UUIDs definitions in nRF Connect

zpm1066, 2024-04-19:
Hello,

Sometime ago, I raised this issue on DevZone, not sure if it got the appropriate attention. Hence, raising the request here.

In nRF Connect for Mobile, custom UUIDs show up as unknown, making it difficult to identify them easily.

Is there a way to add custom UUIDs, as can be done for example on a Mac by adding the custom UUIDs to the "uuid_definitions.json" (located in folder /Users/username/Library/Application\ Support/nrfconnect)?

Thank you.

philips77, 2024-04-19:
We don't have this feature in nRF Connect for iOS yet, but we are aware of your request.
Unfortunately, currently we're busy with some other projects so it will have to wait.

zpm1066, 2024-04-19:
@philips77 Thanks again. Not a problem, I just wanted to make sure the request is in the system and didn’t get lost.

Any idea when the team will get a chance to look at the enhancement request? 3-4 months or longer?

philips77, 2024-04-19:
I'd say longer, but can't say any date, sorry.

dinesharjani, 2024-04-22:
Hi @zpm1066 - there's an alternative solution that works today, that you can use. nRF Connect for Mobile (iOS) pulls all Company, Service, Characteristic and Descriptor definitions from the [Bluetooth Numbers Database](https://github.com/NordicSemiconductor/bluetooth-numbers-database). If you make a pull request to add your UUID definitions there, we can marge them in, and nRF Connect will self-update itself with the new definitions.

zpm1066, 2024-04-22:
@dinesharjani Thank you. I'm aware of this alternative solution. 

However, I'm using the Thingy:53 and would have expected to see the sensor characteristics (temperature, pressure, humidity, red/green/green color, battery voltage and IAQ) appear in readable form rather than "Unknown Characteristic".

On macOS, "uuid_definitions.json" in "/Users/<user>/Library/Application Support/nrfconnect" has the UUIDs with the readable names in English.
 
 [uuid_definitions.json](https://github.com/NordicSemiconductor/IOS-nRF-Connect/files/15066022/uuid_definitions.json)

The "v1" folder in the https://github.com/NordicSemiconductor/bluetooth-numbers-database lists UUIDs for some "Thingy" services but nothing for the above listed Thingy:53 sensors. 
 
I'm sure other Thingy:53 owners have the same issue. Can you update the Blue Numbers Database with the Thingy:53 UUID definitions, as per the Mac/PC uuid_definitions.json? Thanks.

Regards,
Ravi

dinesharjani, 2024-04-23:
So, what you're essentially doing is what I proposed, but locally on the copy nRF Connect downloads to later parse. This is the same thing the iOS version does, but you of course can't access that internal file unless you jailbreak your device.

zpm1066, 2024-04-23:
Understood, but I would have expected the Thingy:53 UUID definitions for the onboard sensors to be already in the Blue Number Database. Purchasers of Thingy:53 shouldn’t be expected to raise a request for the inclusion.

closed 2024-12-12 by dinesharjani