nordicsemi/IOS-nRF-Connect #162: Passkey exchange handling

mpiffari, 2024-06-20:
Hi @dinesharjani @philips77 ,
I am opening this issue to seek your advice or guidelines on handling the pairing procedure programmatically on iOS.

Which is the best way to wait for pairing popup rised by OS?
Do you have any tips or advice that you can share with fellow GitHub users?

Thank you!

dinesharjani, 2024-06-20:
I will close this issue here, since it's not really an issue. But we can have a small discussion here.

So I guess the best way would be to apply "common sense". Pairing procedure, if memory serves, will be initiated on your behalf on iOS whenever you attempt to access an encrypted attribute, for example via Characteristic Read. With this in mind, the user of the app should have some context. The user should initiate, or trigger the pairing request via a button. So for example if you're connecting to a Bluetooth LE device, the user should do that action themselves in the UI. And then, programmatically, you can trigger the call that causes the pairing procedure to start. This way the user gets the feeling that "pairing is happening because I decided to", not because the app is doing stuff they don't know about.

closed 2024-06-20 by dinesharjani

mpiffari, 2024-06-24:
Thank @dinesharjani for your comment.

So, in your personal experience, there is no way to programmatically access "My devices" list (show in Settings > Bluetooth)?

dinesharjani, 2024-06-24:
No, that is absolutely not possible. Apple will never allow that, for privacy reasons amongst others. Otherwise you could know all of the Devices a User owns, such as Headphones, maybe home speakers, Health-related devices, etc. That's a big privacy nightmare.

One thing that does exist is Cross-Transport Key Derivation, I think that's what's called. If the user, not the app, is connected to a Bluetooth (not Bluetooth LE) Device, then you can get a Bluetooth LE connection to that connected Device.