nordicsemi/IOS-nRF-Connect #163: App crashes/restarts when rotating app on DFU screen

jmaha, 2024-07-16:
This bug has been present for a long time. I noticed it's still there in the latest release 2.7.7.

If I do these steps, the app will exhibit a strange almost-crash back to the Scanning screen with all filters reset:

- Connect to a device advertising the Nordic DFU service
- Select a firmware image to install
- On the DFU screen that shows the graphical depiction of how much space the Application/Bootloader will take up, CPU info, and the Start/Cancel buttons, if you rotate the screen (which I frequently do to attempt to read a long filename to make sure I chose the right file), the app restarts without fully crashing: the screen goes black briefly, and then restarts back at the Scanning screen with all filters reset. Very frustrating when I was just trying to double-check a filename before starting DFU.

dinesharjani, 2024-07-16:
I'm very sorry you've been dealing with this issue for these past few months - I wish I'd known sooner. I'll do my best to get this resolved quickly.

Would it help as well if I allowed you to 'not see' the Preview graphic of the contents of your firmware, to give you more space to read the filename, etc?

jmaha, 2024-07-16:
It would definitely be helpful to have some way to see the full filename. It used to be possible to long-press the filename and it would show the whole thing, but that went away in the last UI redesign.

dinesharjani, 2024-07-17:
Hey @jmaha - I've been working on this today. I can't replicate your issue. Connecting to a Nordic DFU Service Device, opening DFU, selecting a firmware file, seeing the Preview, and then rotating multiple times - I found a bug that deselects the "Device Details" after two rotations and doesn't allow you to show Device Details again in Landscape for a big device like the iPhone XR, but I don't see the kind of behaviour you're describing. Could you send me some screenshots / more specific information about the device? iOS version you're running? Or perhaps even a video.

Sorry for the inconvenience.

jmaha, 2024-07-17:
I'm on an iPhone 14 Pro running iOS 17.5.1. 

Here's a video of the crash happening: https://youtube.com/shorts/e5ayAYfW6o4

jmaha, 2024-07-17:
The filename of the OTA image is 49 characters. This image has both an application and bootloader -- it's an image based on the nRF5 SDK.

dinesharjani, 2024-07-17:
Okay, this helps. I guess it might be that you left the filter open, I did not try this use case. It's also one of the iPhone(s) with the Capsule instead of the "Notch" (I call them Batman ears) which I don't have to test, but I'll see what I can do. Thanks!

dinesharjani, 2024-07-18:
Just an update - I'm working as diligently as I can on the issue. Trying to find an accurate way to reproduce it, since I have a stack trace of it but it's triggered by system calls due to the rotation event, so I don't get a location on our code of where source of the issue is.

jmaha, 2024-07-25:
I see. Let me know if I can help gather any information.

dinesharjani, 2024-07-31:
Hello @jmaha - I made a lot of underlying changes to the app, since I can't actually reproduce the issue in any of my devices. I think it's likely the crash will still occur, but if it does, please enable the "Send Feedback" option Apple provides when installing the new beta. Here's the link for you to test this out: https://testflight.apple.com/join/FwMQS8tH

jmaha, 2024-07-31:
Hi @dinesharjani, I just installed the TestFlight version and indeed the app does still crash when I rotate after selecting a DFU image. I submitted beta feedback via the testing mechanism when it reported the crash. Hopefully the stack trace arrives shortly!

dinesharjani, 2024-08-02:
Hi @jmaha - a crash on rotation was fixed. However, it did not match the reported stack trace. Still I'd appreciate if you could try it out again (you should get a TestFlight notification to update or open TestFlight and tap Update) and if it still happens, to please attach a video once again. Also, a couple of details if you could provide them: does this happen straight after application launch, or only after performing other operations such as DFU? Does it happen if the Scanner Filter is closed instead of open? Does it happen (the crash) if you rotate before selecting a DFU package and you're in a different tab?

Thanks.

jmaha, 2024-08-02:
I just tested the updated app, and it still crashes the same way. I tested both with and without the scan filter. I submitted several crash reports/stack traces. Video:

https://youtube.com/shorts/F6tc115P9pQ?feature=share￼
Crash with nRF Connect 2.7.8 build 83
youtube.com

dinesharjani, 2024-08-03:
Okay. I'm going to try something out on Monday. Something tells me that'll be it, but let's keep our fingers crossed.

Thank you for your valuable time.

dinesharjani, 2024-08-05:
Hey @jmaha - there's a new build, number 92, out. Please try it out when you can. Thanks.

jmaha, 2024-08-05:
Hi Dinesh,

I just tried the new build and the DFU screen still crashes on rotation. However, I did notice that in portrait orientation the firmware image name now wraps to the next line, so I’m able to read the entire name, thus eliminating the need for rotation.

That said, I’m happy to continue to test this for you if you’re going to keep chasing this crash!

Best,
Jason

dinesharjani, 2024-08-09:
Hi Jason,

We might have to give it a rest, yes. Anyway, there's a new update in TestFlight now. Could you try that anyway for me, please?

dinesharjani, 2024-08-29:
Update: I replicated the crash.

closed 2024-12-12 by dinesharjani