nordicsemi/IOS-nRF-Connect #164: Unable to edit advertiser services

tomschmitz98, 2024-07-30:
IOS Version 17.5.1 (Build 21F90)
Device Model: iPhone 14 Pro
nRF Connect version: 2.7.7 Build 165

Steps to reproduce the bug:
1. Under Peripheral, click "Add Advertiser"
2. Rename advertiser to something other than iPhone
3. Click "Add Serice"
4. Attempt to change service UUID from 181C to 180D and press "save" (It won't let you save the service. Also tried looking up service by service name and it still won't let me save)

dinesharjani, 2024-07-31:
This is correct. I just fixed it yesterday for version 2.7.8. I think I checked it in to the latest TestFlight Beta, which you can try out via this link: https://testflight.apple.com/join/FwMQS8tH

Please report back when you can :)

tomschmitz98, 2024-07-31:
Thanks! It looks like the bug is resolved :)

closed 2024-07-31 by tomschmitz98

dinesharjani, 2024-07-31:
Happy to hear it. Thanks for reporting.