nordicsemi/IOS-nRF-Connect #167: Allow temporary hide of items in RSSI display/ allow subtraction of list of earlier devices detected [enhancement]

gaitskell, 2024-11-23:
Suggested enhancements

Would it be possible to hide some devices from active list? There are so many Bluetooth devices in the environment now it would help when studying a subset/individual device to be able to temporarily hide some of them based on simply long press or swipe left in list?

it would also be great to be able to save internally a snapshot/current list of devices and use that to mask future display. E.g. so only those devices are shown in graph or in list in future, Or so only new devices not in the early snapshot  are shown/graohed.

thanks for a great app, appreciate all the work

dinesharjani, 2024-11-25:
hi @gaitskell - from the Scanner, at the top just below "Scanner" you can read "No Filter". If you tap that, Filtering options open up and you can use them to remove items from the list.

Saving a "snapshot" of the currently scanned list, there is code for that, but it is not currently implemented as a user-level feature. We could do that in the future.

Thanks for the feedback!

closed 2025-01-21 by dinesharjani

DG12, 2025-01-22:
> ID: <NordicSemiconductor/IOS-nRF-Connect/issue/167/issue_event/16013876421
> @github.com>

I am no longer able to download test versions  via
appdistribution.firebase.google.com
Can this be reinstated?
Thank you,
Dennis German
***@***.***
aka ***@***.***