nordicsemi/IOS-nRF-Connect #169: Thingy:53 - Request to add IAQ sensor UUID so that the "Unknown Characteristic" shows up with "IAQ" in nRF Connect for Mobile

zpm1066, 2024-12-11:
Hello,

A while back, through a Nordic Support issue, I requested UUIDs to be added to Thingy:53 so that the sensor characteristics appear with names rather than "unknown". From my understanding, the nRF Connect for Mobile dev team made the required changes. However, the IAQ sensor UUID entry didn't get included as it was one missed from the original Nordic Semi 'Thingy53 sensor hub' blog.

https://devzone.nordicsemi.com/nordic/nordic-blog/b/blog/posts/thingy-53-sensor-hub

[image]

```
    { "name": "Thingy:53 Sensor Hub Service", "identifier": "com.nordicsemi.service.thingy53_sensor_hub", "uuid": "A5B46352-9D13-479F-9FCB-3DCDF0A13F4D" , "source": "nordic"},
    { "name": "Thingy:53 Sensor Hub Temp", "identifier": "com.nordicsemi.service.thingy53_temp", "uuid": "506A55C4-B5E7-46FA-8326-8ACAEB1189EB" , "source": "nordic"},
    { "name": "Thingy:53 Sensor Hub Pressure", "identifier": "com.nordicsemi.service.thingy53_pressure", "uuid": "51838AFF-2D9A-B32A-B32A-8187E41664BA" , "source": "nordic"},
    { "name": "Thingy:53 Sensor Hub Humidity", "identifier": "com.nordicsemi.service.thingy53_humidity", "uuid": "753E3050-DF06-49334B53-B090-5E1D810C4383" , "source": "nordic"},
    { "name": "Thingy:53 Sensor Hub Red Color", "identifier": "com.nordicsemi.service.thingy53_red_color", "uuid": "82754BBB-6ED3-4D69-A0E1-F19F6B654EC2"  , "source": "nordic"},
    { "name": "Thingy:53 Sensor Hub Green Color", "identifier": "com.nordicsemi.service.thingy53_green_color", "uuid": "DB7F9F36-92CE-4509-A2EF-AF72BA38FB48" , "source": "nordic"},
    { "name": "Thingy:53 Sensor Hub Blue Color", "identifier": "com.nordicsemi.service.thingy53_blue_color", "uuid": "F5D2EAB5-41E8-4F7C-AEF7-C9FFF4C544C0" , "source": "nordic"},
    { "name": "Thingy:53 Sensor Hub Battery Voltage", "identifier": "com.nordicsemi.service.thingy53_battery_voltage", "uuid": "FA3CF070-D0C7-4668-96C4-86125C8AC5DF" , "source": "nordic"},
    { "name": "Thingy:53 Sensor Hub IAQ", "identifier": "com.nordicsemi.service.thingy53_iaq", "uuid": "AD79CBCE-65DF-4BC6-BDD4-8CE2B6B75C59" , "source": "nordic"},

```

As you can see, I'm currently using a random UUID for IAQ.

Could you please generate a Nordic Semi UUID for the IAQ sensor so that customers can utilize it? Thank you.

The BME688 sensor with air quality IAQ is in the **Thing:53** and now also in the new **Thingy:91X**. This request would be beneficial for all customers.

Regards,
Ravi

dinesharjani, 2025-01-07:
If memory serves, these were added. Please reopen if further work is required.

closed 2025-01-07 by dinesharjani

zpm1066, 2025-01-08:
@dinesharjani 
I don't quite know why this issue has been closed. 
I had created a new issue to request addition of IAQ sensor UUID as per text above. All the other Thingy:53 sensor UUIDs were update so that they do not show "Unknown Characteristic" in nRF Connect for Mobile app.

Please add a Nordic Semi generated UUID for the IAQ sensor, so that customers won't have to use a random UUID for the IAQ sensor on Thingy:53 and Thingy:91X.

Thank you.

zpm1066, 2025-01-08:
I don't see an option to re-open the issue.

dinesharjani, 2025-01-08:
As far as I know, Nordic has not generated a UUID for IAQ Sensor. So there is no definitive UUID to tie-in here. We're in a bit of a crossroads.

zpm1066, 2025-01-08:
Surely there must be a process within Nordic Dev on how to allocate a new unique UUID.

zpm1066, 2025-01-15:
OK, looks like we'll see this progress to a resolution.

I've been working with Nordic Support and the local RSM regarding this request. Support have kindly informed that they'll get the issue resolved. 

A Nordic UUID for the IAQ sensor will get generated and added to Tiago Monte "Thingy:53 Sensor Hub" blogpost , and then get supported in the app.

zpm1066, 2025-01-21:
With Nordic Support's assistance, this issue has now been resolved. They have updated the NordiSemi bluetooth characteristics UUID json with an entry for IAQ, as per original request here.

dinesharjani, 2025-01-22:
And it was me who did the update ;)

zpm1066, 2025-01-22:
@dinesharjani , Thank you. It’s much appreciated.