nordicsemi/IOS-nRF-Connect #170: nRF Connect for Mobile - Request for an enhancement to include option to select the data parser default for the application

zpm1066, 2024-12-11:
Hello,

Currently, there is no option to globally select a data parser for the nRF Connect for Mobile app. 

The data parser is fixed in code to be "Byte Array (Hex)' at the application level. A configurable option to select the data parser would aid usability, e.g. selecting "Unsigned int (8, 16, or 32)" to display value sin decimal.

Please consider adding a configurable option for "Set Data Parser". Thank you.

Regards,
Ravi

dinesharjani, 2024-12-12:
There is a button that looks like double quotes "" next to the copy button that allows you to set the Data Parser.

closed 2024-12-12 by dinesharjani

zpm1066, 2024-12-12:
Yes, the button can be used to set the data parser. However, the change only persists for the session. There isn’t a way to make it a default. Currently the default for the data parser is hex, like to make decimal tge default.