nordicsemi/IOS-nRF-Connect #172: DFU: Variable connection interval and the app

reddwarfSiddis, 2025-01-21:
Hi,
App version 2.78.14.
We are developing an app and using the DFU part of the app extensively. What we have seen is that approximately every 5 second the connection interval goes from 18.5ms (good for DFU transfer) to over 180ms. The connection interval will after some seconds again go back to 18.5ms. This influences the transfer of data, and the data rate is varying. We do not see this on Android.
I found some configs in the SDK that has affected the connection interval.
The configs are
CONFIG_MCUMGR_TRANSPORT_BT_CONN_PARAM_CONTROL=y
CONFIG_MCUMGR_TRANSPORT_BT_CONN_PARAM_CONTROL_MIN_INT=6
CONFIG_MCUMGR_TRANSPORT_BT_CONN_PARAM_CONTROL_MAX_INT=9
CONFIG_MCUMGR_TRANSPORT_BT_CONN_PARAM_CONTROL_LATENCY=0
CONFIG_MCUMGR_TRANSPORT_BT_CONN_PARAM_CONTROL_TIMEOUT=42
CONFIG_MCUMGR_TRANSPORT_BT_CONN_PARAM_CONTROL_RESTORE_TIME=5000
CONFIG_MCUMGR_TRANSPORT_BT_CONN_PARAM_CONTROL_RETRY_TIME=1000

I did a trial, and changed CONFIG_MCUMGR_TRANSPORT_BT_CONN_PARAM_CONTROL_RESTORE_TIME to 65535. I now see that the the connection interval does not vary (stays and 18.5ms), but after approximately 5 seconds the app almost stalls; no data gets transmitted. 
I have to cancel the transfer and try over again, but then after another 5 seconds the transfer stops again.

dinesharjani, 2025-01-21:
Wow, this is great feedback. I can try to reproduce this today and see if there's a bottleneck in the code, which is possible. Can you tell me which chip / board did you build this configuration for? Was it 52840?

dinesharjani, 2025-01-27:
Hi there,

I worked hard over the last few (working days) to try to find the source of the issue. I can't know for sure if I did fix it or not, but I did fix a cause that has very similar effects as you described, a five second stall and so on. Keep in mind that the BLE radio works very differently between Android and iOS - on iOS all apps share physical connections to the same Device, and are kept in the dark about it via the API. We also have no control about the radio timings, and iOS apps share the BLE Radio with Apple's own uses for it, such as audio, device tracking, and so on. Having such low connection intervals means iOS will also expect timely responses on said intervals, and we've also seen in the past 5340s having tight loops on the BLE Antenna thread missing those transmissions. For increased speed, I'd instead recommend increasing MTU size, Number of Buffers for parallel writes and the Reassembly Buffer Size. This also increases speed on Android. 

The aforementioned improvements are already a part of the https://github.com/NordicSemiconductor/IOS-nRF-Connect-Device-Manager library, but have not been released in the app yet. Version 2.7.15 does not include it yet.

closed 2025-02-10 by dinesharjani