nordicsemi/IOS-nRF-Connect #174: Is there a difference between Android write and iOS write?

tobiyas09, 2025-02-05:
I have a Bluetooth Smart Card Reader (ISO15693) that I use to read and write NFC cards. I am working on a React native mobile app that uses [react-native-ble-manager](https://github.com/innoveit/react-native-ble-manager) for Bluetooth stuff. On Android, everything works perfectly. However, on iOS—especially on iPad—the connection sometimes hangs and the device disconnects when writing data.

I thought that something was wrong with my code, or in [react-native-ble-manager](https://github.com/innoveit/react-native-ble-manager). But then I tried to write a simple command using the nRF-Connect app on iPad and the same thing happened.

I connect to the device, turn on the notifications, and send the byte array data 006202. Sometimes I do get the value back, but most of the time I get nothing, the writing is stuck and the device disconnects.

Do you have any idea why this happens?
The problem is probably in the smart card reader

dinesharjani, 2025-02-06:
Hi,

Hmmm. We don't have much context here about what could be happening other than the differences you mention between Android and iOS. On iOS, if you nRF Connect, you can switch over to the Logs Tab, turn on all Log Levels and check if we get the callback after performing the write. Also, you may try writeWithResponse or writeWithoutResponse. The best answer is probably to use a BLE Sniffer and compare packet traffic between iOS and Android. [Apple allows iPhone(s) to be used as sniffers on a Mac](https://www.bluetooth.com/blog/a-new-way-to-debug-iosbluetooth-applications/). You may try that.

tobiyas09, 2025-02-06:
Hi, thanks for the reply I will check the sniffer. The logs are not helpful, because the device disconnects without error.
This is the successful case, when I get the value back.
```
[12:35:26.6520] Verbose: Connecting...
[12:35:26.6520] Debug: cbCentralManager.connect()
[12:35:26.9930] Debug: [Callback] centralManager(central, didConnect: peripheral)
[12:35:26.9930] Normal: Connected.
[12:35:26.9930] Verbose: Discovering Services...
[12:35:26.9930] Debug: peripheral.discoverServices(nil)
[12:35:27.3490] Debug: [Callback] peripheral(peripheral, didDiscoverServices: nil)
[12:35:27.3490] Normal: Discovered FFF0 Services.
[12:35:27.3500] Verbose: Discovering Characteristics for FFF0...
[12:35:27.3500] Debug: peripheral.discoverCharacteristics(nil, for: FFF0)
[12:35:27.4700] Debug: [Callback] peripheral(peripheral, didDiscoverCharacteristicsFor: FFF0, error: nil)
[12:35:27.4700] Verbose: Discovering Descriptors for Characteristic FFF1...
[12:35:27.4700] Debug: peripheral.discoverDescriptors(for: FFF1)
[12:35:27.4700] Verbose: Discovering Descriptors for Characteristic FFF2...
[12:35:27.4710] Debug: peripheral.discoverDescriptors(for: FFF2)
[12:35:27.4710] Debug: [Callback] peripheral(peripheral, didDiscoverDescriptorsFor: FFF1, error: nil)
[12:35:27.4730] Normal: Discovered FFF1 and FFF2 Characteristics for Service FFF0.
[12:35:27.4740] Normal: FFF1 has no Descriptors.
[12:35:27.5900] Debug: [Callback] peripheral(peripheral, didDiscoverDescriptorsFor: FFF2, error: nil)
[12:35:27.5930] Normal: Discovered Client Characteristic Configuration Descriptors for Characteristic FFF2
[12:35:33.1190] Verbose: Writing '' to Characteristic FFF2
[12:35:33.1190] Debug: peripheral.writeValue(, forCharacteristic: FFF2)
[12:35:33.1190] Application: Writing value '' to FFF2 Characteristic
[12:35:33.1730] Debug: [Callback] peripheral(peripheral, didWriteValueForCharacteristic: FFF2, error: nil)
[12:35:33.1730] Normal: Wrote 0 bytes to FFF2 Characteristic
[12:35:33.1730] Application: Wrote '' to FFF2 Characteristic
[12:35:34.0840] Verbose: Writing '01' to Characteristic FFF2
[12:35:34.0840] Debug: peripheral.setNotifyValue(01, forCharacteristic: FFF2)
[12:35:34.0850] Application: Setting Boolean true for Notifying Characteristic FFF2
[12:35:34.1330] Debug: [Callback] peripheral(peripheral, didUpdateNotificationStateFor: FFF2, error: nil)
[12:35:41.6080] Verbose: Writing '0062 02' to Characteristic FFF2
[12:35:41.6080] Debug: peripheral.writeValue(0062 02, forCharacteristic: FFF2)
[12:35:41.6090] Application: Writing value '0062 02' to FFF2 Characteristic
[12:35:41.6330] Debug: [Callback] peripheral(peripheral, didWriteValueForCharacteristic: FFF2, error: nil)
[12:35:41.6330] Normal: Wrote 006202 to FFF2 Characteristic
[12:35:41.6330] Application: Wrote '0062 02' to FFF2 Characteristic
[12:35:41.6530] Debug: [Callback] peripheral(peripheral, didUpdateValueForCharacteristic: FFF2, error: nil)
[12:35:41.6530] Normal: Updated Value of Characteristic FFF2 to 0063 9000 0413 3527 0C08.
[12:35:41.6530] Application: "0063 9000 0413 3527 0C08" value received.
[12:35:53.5660] Verbose: Writing '00' to Characteristic FFF2
[12:35:53.5660] Debug: peripheral.setNotifyValue(00, forCharacteristic: FFF2)
[12:35:53.5670] Application: Setting Boolean false for Notifying Characteristic FFF2
[12:35:53.5930] Debug: [Callback] peripheral(peripheral, didUpdateNotificationStateFor: FFF2, error: nil)
[12:35:54.6990] Verbose: Writing '01' to Characteristic FFF2
[12:35:54.6990] Debug: peripheral.setNotifyValue(01, forCharacteristic: FFF2)
[12:35:54.7000] Application: Setting Boolean true for Notifying Characteristic FFF2
[12:35:54.7330] Debug: [Callback] peripheral(peripheral, didUpdateNotificationStateFor: FFF2, error: nil)
[12:36:00.2680] Verbose: Writing '0062 02' to Characteristic FFF2
[12:36:00.2680] Debug: peripheral.writeValue(0062 02, forCharacteristic: FFF2)
[12:36:00.2690] Application: Writing value '0062 02' to FFF2 Characteristic
[12:36:00.3330] Debug: [Callback] peripheral(peripheral, didWriteValueForCharacteristic: FFF2, error: nil)
[12:36:00.3330] Normal: Wrote 006202 to FFF2 Characteristic
[12:36:00.3330] Application: Wrote '0062 02' to FFF2 Characteristic
[12:36:00.3530] Debug: [Callback] peripheral(peripheral, didUpdateValueForCharacteristic: FFF2, error: nil)
[12:36:00.3530] Normal: Updated Value of Characteristic FFF2 to 0063 9000 0413 3527 0C08.
[12:36:00.3530] Application: "0063 9000 0413 3527 0C08" value received.
```
Here the device disconnects.
```
[12:37:01.6920] Verbose: Writing '01' to Characteristic FFF2
[12:37:01.6920] Debug: peripheral.setNotifyValue(01, forCharacteristic: FFF2)
[12:37:01.6930] Application: Setting Boolean true for Notifying Characteristic FFF2
[12:37:07.6850] Verbose: Writing '0062 02' to Characteristic FFF2
[12:37:07.6850] Debug: peripheral.writeValue(0062 02, forCharacteristic: FFF2)
[12:37:07.6850] Application: Writing value '0062 02' to FFF2 Characteristic
[12:37:18.8880] Normal: Scanner Off.
[12:37:31.7540] Debug: [Callback] centralManager(central, didDisconnectPeripheral: peripheral, error: nil)
[12:37:31.7540] Normal: Disconnected.
```

dinesharjani, 2025-02-06:
I suspect something is going on at a lower level than we get via iOS APIs. So, the sniffer should provide you with more answers.