nordicsemi/IOS-nRF-Connect #178: Question about Using this App on iOS

ankumar8164, 2025-03-20:
This is not exactly an issue, more of a question. I am trying to do some work with an ESP32S3, following some of the instructions here:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/ble/get-started/ble-introduction.html

Specifically, I am trying to do some work with the GATT server, using the NimBLE framework. However, the application shown in the tutorial seems to be very different from the one on my iOS phone. I do not seem able to see any services (or perhaps I do not know where to look). Is the tutorial outdated? Is there a more up-to-date tutorial I can look at for how to use this application?

dinesharjani, 2025-03-21:
Hi,

The app shown in the tutorial is the Android version of nRF Connect. They look a bit different owing to their different platforms :)

On iOS it's similar. On the Scanner Tab find your nimBLE device. Connect to it, swipe right to see the Client / Connected Services, and scroll to the correct Service, then tap the "Up" arrow to write the bytes to turn Off / On the LED.

closed 2025-03-21 by dinesharjani

ankumar8164, 2025-03-21:
Got it, thank you for your help! I'm trying to follow those steps you mentioned for iOS, but I don't seem to see Client / Connected Services anywhere. These are the different tabs I see:

[image]

[image]

[image]

[image]

ankumar8164, 2025-03-21:
Oh also, I have this message: "We're sorry, but this device does not support Nordic or McuMgr DFU update mechanisms" on my phone.

ankumar8164, 2025-03-21:
Thank you so much! That fixed it for me.