nordicsemi/IOS-nRF-Connect #186: Reading Characteristic Results in Vibrating until disconnecting

MB-Conti, 2025-04-06:
Hi,
I am using nrf Connect for iOS on version 2.7.17 on iOS 18.4 Build 22E240 on a iPhone SE.

If I try to read a characteristic my phone starts to vibrate until I end the connection. This happens if I connect to my own device but also to any device of different manufacturers.

Anyone else who experience the same issue?

MB-Conti, 2025-04-07:
Hi Dinesh,
I don't think this is the problem. I so far only worked with the Android version of the app, so the handling of the iOS version I still need to adapt. However, the problem also occurs reading for instance the characteristics of the Device Information Service, which I believer should not have the possibility to enable notifications

dinesharjani, 2025-04-08:
Can you share a screenshot?

Tanut462, 2025-12-04:
Thai