nordicsemi/IOS-nRF-Connect #192: Unable to find the BLE devices in my iOS app along with nrf connect app?

vadlani9, 2025-08-11:
We are using 2 names for my IoT devices, while pairing/adding ble advertisement name is **'BO-BleDev'** and change the network **'CA_'BO-BleDev'**. 

Here while pairing it's visible always but while changing the device network it's not visible in the scan in both my iOS app and nrf connect app. But for some iPhone devices it's found in BLE scan but not all iPhones.

But same time in Android it's found in ble scan. 

Any additional information required I can provide.

philips77, 2025-08-11:
Hello,\
I believe the reason os the last "Error with parsing EIR (0x02)". Remove the last byte from the adv data of the device. iPhones may be rejecting the packets due to that.

vadlani9, 2025-08-11:
ok, Thank you we will check and let you know.
And any other reasons for this issue.